hbldh/pymetawear #3: Notifications?

siddrajput, 2016-04-01:
I want to read the MetaEear switch state as a notification.  When the switch is pressed I want to be notified.  Is it supported?

hbldh, 2016-04-02:
I have had some problems getting notifications to work with pygattlib which the current version pymetawear is using. I have another solution working with the pygatt package instead, but it needs some more testing before I merge it to master. Maybe done on Monday.

Try it out in the develop branch if you feel like it, but there is no documentation for it yet.

hbldh, 2016-04-03:
@siddrajput You can use the new 0.2.0 release with pygatt backend (see [LED example](https://github.com/hbldh/pymetawear/blob/master/examples/led.py) for code).

You can then make a subscription like this:

```python

def switch_notification_handler(handle, value):
    print("Notify 0x{0:04x}: {1}".format(
        handle, " ".join(["{:02x}".format(b) for b in value])))

c.requester.subscribe(str(uuid_to_switch_char), switch_notification_handler)

```

I have not tested the switch yet and I don't know its characteristic uuid and it might be prudent to use some libmetawear function to create the subscription or at least get the charectersitic, but I don't know how yet.

I will add an example for it in the new [examples](https://github.com/hbldh/pymetawear/tree/master/examples) folder when I have succeeded. If you get around to it faster than me, please let me know!

hbldh, 2016-04-04:
The subscribing to switch notifications was simpler than I imagined! Take a look at the [switch example code](https://github.com/hbldh/pymetawear/blob/master/examples/switch.py) to see how a subscription is done, by using exclusively C++ API methods.

It works fine for me, on both a Linux Mint 17 installation and a Ubuntu 15.10 one, using both a R Pro board and a C.

Hope you are able to try it out!

closed 2016-04-09 by hbldh