hbldh/pymetawear #10: gattlib error

siddrajput, 2016-04-14:
sudo pip install gattlib gives error:

``edison@ubilinux:/home/pymetawear$ sudo pip install gattlib
Downloading/unpacking gattlib
  Downloading gattlib-0.20150805.tar.gz (1.7Mb): 1.7Mb downloaded
  Running setup.py egg_info for package gattlib

Installing collected packages: gattlib
  Running setup.py install for gattlib
    building 'gattlib' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/gattservices.cpp -o build/temp.linux-i686-2.7/src/gattservices.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/beacon.cpp -o build/temp.linux-i686-2.7/src/beacon.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bindings.cpp -o build/temp.linux-i686-2.7/src/bindings.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/gattlib.cpp -o build/temp.linux-i686-2.7/src/gattlib.o
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/lib/uuid.c -o build/temp.linux-i686-2.7/src/bluez/lib/uuid.o
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/gatt.c -o build/temp.linux-i686-2.7/src/bluez/attrib/gatt.o
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/gattrib.c -o build/temp.linux-i686-2.7/src/bluez/attrib/gattrib.o
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/utils.c -o build/temp.linux-i686-2.7/src/bluez/attrib/utils.o
    src/bluez/attrib/utils.c: In function ‘gatt_connect’:
    src/bluez/attrib/utils.c:66:15: error: ‘BDADDR_LE_RANDOM’ undeclared (first use in this function)
    src/bluez/attrib/utils.c:66:15: note: each undeclared identifier is reported only once for each function it appears in
    src/bluez/attrib/utils.c:68:15: error: ‘BDADDR_LE_PUBLIC’ undeclared (first use in this function)
    error: command 'gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools;__file__='/home/pymetawear/build/gattlib/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-APzINk-record/install-record.txt:
    running install

running build

running build_ext

building 'gattlib' extension

creating build

creating build/temp.linux-i686-2.7

creating build/temp.linux-i686-2.7/src

creating build/temp.linux-i686-2.7/src/bluez

creating build/temp.linux-i686-2.7/src/bluez/lib

creating build/temp.linux-i686-2.7/src/bluez/attrib

creating build/temp.linux-i686-2.7/src/bluez/src

creating build/temp.linux-i686-2.7/src/bluez/src/shared

creating build/temp.linux-i686-2.7/src/bluez/btio

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/gattservices.cpp -o build/temp.linux-i686-2.7/src/gattservices.o

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/beacon.cpp -o build/temp.linux-i686-2.7/src/beacon.o

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bindings.cpp -o build/temp.linux-i686-2.7/src/bindings.o

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/gattlib.cpp -o build/temp.linux-i686-2.7/src/gattlib.o

cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++ [enabled by default]

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/lib/uuid.c -o build/temp.linux-i686-2.7/src/bluez/lib/uuid.o

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/gatt.c -o build/temp.linux-i686-2.7/src/bluez/attrib/gatt.o

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/gattrib.c -o build/temp.linux-i686-2.7/src/bluez/attrib/gattrib.o

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -DVERSION="5.25" -I/usr/include/glib-2.0 -I/usr/lib/i386-linux-gnu/glib-2.0/include -Isrc/bluez -I/usr/include/python2.7 -c src/bluez/attrib/utils.c -o build/temp.linux-i686-2.7/src/bluez/attrib/utils.o

src/bluez/attrib/utils.c: In function ‘gatt_connect’:

src/bluez/attrib/utils.c:66:15: error: ‘BDADDR_LE_RANDOM’ undeclared (first use in this function)

src/bluez/attrib/utils.c:66:15: note: each undeclared identifier is reported only once for each function it appears in

src/bluez/attrib/utils.c:68:15: error: ‘BDADDR_LE_PUBLIC’ undeclared (first use in this function)

error: command 'gcc' failed with exit status 1

----------------------------------------
Command /usr/bin/python -c "import setuptools;__file__='/home/pymetawear/build/gattlib/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-APzINk-record/install-record.txt failed with error code 1 in /home/pymetawear/build/gattlib
Storing complete log in /root/.pip/pip.log
edison@ubilinux:/home/pymetawear$

[pip.zip](https://github.com/hbldh/pymetawear/files/218602/pip.zip)

hbldh, 2016-04-14:
Have you got all requirements installed, most notably `libbluetooth-dev` and that it is of version 4.101 or greater?

What operating system are you using? If it is some solution running on ARM architecture, checkout these issues from the `gattlib` library instead:

https://bitbucket.org/OscarAcena/pygattlib/issues/34/error-on-raspberry-pi-2
https://bitbucket.org/OscarAcena/pygattlib/issues/21/build-error-bdaddr_le_random-undeclared

I have not had time to test `pymetawear` on anything else than Ubuntu and Linux Mint as of yet.

hbldh, 2016-04-14:
I installed a Vagrant box with Debian Wheezy (base dist of the Ubilinux that I think you are using) and by running this:

```bash
sudo apt-get install libbluetooth-dev
dpkg --status libbluetooth-dev | grep ^Version:
```
I get the result
```
Version: 4.99-2
```

* `gattlib` backend requires [4.101 or greater](https://bitbucket.org/OscarAcena/pygattlib/src/a858e8626a93cb9b4ad56f3fb980a6517a0702c6/DEPENDS)
* `pygatt` with the `gatttool` solution that I employ says it requires Bluez/`libbluetooth-dev` version of [5.18 or greater](https://github.com/peplin/pygatt#requirements). I have gotten it to work with 4.101, though.

In short, none of the Python packages for Bluetooth Low Energy communication that I have found works with `wheezy`'s package version.

siddrajput, 2016-04-14:
I have version 4.99-2

```
edison@ubilinux:/home/pymetawear$ dpkg --status libbluetooth-dev | grep ^Version:
Version: 4.99-2
```

hbldh, 2016-04-15:
None of the Python Gatt libraries say that they support that version. 
The PyGatt library might work with the modifications that are present in https://github.com/hbldh/pymetawear/blob/master/pymetawear/backends/pygatt/gatttool.py, but I have no time to verify that right now; you have to explore that yourself. I have gotten it to work with the changes included in PyMetaWear using Bluez 4.101, even though PyGatt says they require 5.X.

The node.js lib [Noble](https://github.com/sandeepmistry/noble/wiki/Getting-started) (used by the [node-metawear](https://github.com/brainexe/node-metawear) library) supports 4.99. It might be a good idea to check that one out.

closed 2016-04-19 by hbldh