hbldh/pymetawear #7: LED.py doesn't work

siddrajput, 2016-04-14:
I'm getting the following message when I run LED.py

`edison@ubilinux:/home/pymetawear/examples$ sudo python led.py
Traceback (most recent call last):
  File "led.py", line 21, in <module>
    from pymetawear.client import discover_devices, MetaWearClient, libmetawear
  File "/home/pymetawear/pymetawear/client.py", line 30, in <module>
    from pymetawear.backends.pygatt import PyGattBackend
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 22, in <module>
    from pymetawear.backends.pygatt.gatttool import PyMetaWearGATTToolBackend
  File "/home/pymetawear/pymetawear/backends/pygatt/gatttool.py", line 18, in <module>
    import pygatt
  File "/usr/local/lib/python2.7/dist-packages/pygatt/__init__.py", line 3, in <module>
    from .backends import BGAPIBackend, GATTToolBackend  # noqa
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/__init__.py", line 2, in <module>
    from .bgapi.bgapi import BGAPIBackend  # noqa
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/bgapi/bgapi.py", line 9, in <module>
    import serial
ImportError: No module named serial
edison@ubilinux:/home/pymetawear/examples$
`

hbldh, 2016-04-14:
Seems like you have missed to install the `pyserial`; that should have been done automatically when installing the library...

Try running
```bash
pip install -U pygatt[GATTTOOL] --no-cache-dir
```
to reinstall the `pygatt` package.

hbldh, 2016-04-14:
Or just install `serial` by

```bash
pip install -U pyserial
```

siddrajput, 2016-04-14:
Ok.  I installed pyserial, gattlib, and updated bluez.  Now I'm getting the following error:

`edison@ubilinux:/home/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTToolBackend and starting GATTtool process...
Connecting GATTTool...
No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "led.py", line 32, in <module>
    c = MetaWearClient(str(address), debug=True)
  File "/home/pymetawear/pymetawear/client.py", line 110, in __init__
    self._address, debug=debug)
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 30, in __init__
    super(PyGattBackend, self).__init__(address, async, debug)
  File "/home/pymetawear/pymetawear/backends/__init__.py", line 53, in __init__
    METAWEAR_SERVICE_NOTIFY_CHAR[1])
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 104, in get_handle
    return self.requester.get_handle(uuid) + int(notify_handle)
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 49, in requester
    self._address, timeout=10.0, address_type='random')
  File "/home/pymetawear/pymetawear/backends/pygatt/gatttool.py", line 60, in connect
    raise NotConnectedError(message)
pygatt.exceptions.NotConnectedError: <NotConnectedError Timed out connecting to CA:CD:4F:09:A8:DE after 10.0 seconds.>
edison@ubilinux:/home/pymetawear/examples$
`

hbldh, 2016-04-14:
Try to change [this_row](https://github.com/hbldh/pymetawear/blob/master/examples/led.py#L32) to this:

```python
c = MetaWearClient(str(address), 'pybluez', debug=True)
```

and see if that can connect. The `pygatt` library tries to parse the gatttool interactive terminal's output, and I have not tested it thoroughly yet.

siddrajput, 2016-04-14:
After changing to:

`c = MetaWearClient(str(address), 'pybluez', debug=True)`

Now I get this:

`edison@ubilinux:/home/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTRequester...
Connecting GATTRequester...
Subscribe 0x001f:
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/pymetawear/pymetawear/backends/__init__.py", line 106, in mbl_mw_read_gatt_char
    response = self.read_gatt_char_by_uuid(characteristic_uuid)
  File "/home/pymetawear/pymetawear/backends/pybluez/__init__.py", line 109, in read_gatt_char_by_uuid
    return self.requester.read_by_uuid(str(characteristic_uuid))[0]
RuntimeError: Characteristic value/descriptor operation failed: No attribute found within the given range
Waiting for MetaWear board to be fully initialized...
`

hbldh, 2016-04-15:
Is this on the same machine running BlueZ 4.99? Your safest bet, as I said in #10, is to go with `pygatt` backend and try to modify [here](https://github.com/hbldh/pymetawear/blob/master/pymetawear/backends/pygatt/gatttool.py#L89-L90) to make pygatt recognise the login in gatttool.

Speaking of gatttool, can you connect to the MetaWear board using the interactive prompt directly:

```bash
hbldh@devbox ~/Repos/pymetawear $ gatttool -I
[   ][                 ][LE]> connect CA:CD:4F:09:A8:DE random
[CON][CA:CD:4F:09:A8:DE][LE]> 
```

If this can be done, then the [pexpect code](https://github.com/hbldh/pymetawear/blob/master/pymetawear/backends/pygatt/gatttool.py#L89-L90) of pygatt just fails to recognize the login, and you will have to find out why.

siddrajput, 2016-04-17:
My login is slightly different.  There is no [CON]

```
edison@ubilinux:/home/pymetawear/examples$ sudo gatttool -I
[                 ][LE]> connect CA:CD:4F:09:A8:DE random
Attempting to connect to CA:CD:4F:09:A8:DE
Connection successful
[CA:CD:4F:09:A8:DE][LE]>
```

So I changed gatttool.py line 53 as:
```
if int(major) < 5:
                    self._con.expect(b'\[{0}\]\[LE\]>'.format(self._address), timeout)
```

siddrajput, 2016-04-17:
Now I'm getting:

`
edison@ubilinux:/home/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTToolBackend and starting GATTtool process...
Connecting GATTTool...
No handlers could be found for logger "pygatt.device"
Traceback (most recent call last):
  File "led.py", line 32, in <module>
    c = MetaWearClient(str(address), debug=True)
  File "/home/pymetawear/pymetawear/client.py", line 110, in __init__
    self._address, debug=debug)
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 30, in __init__
    super(PyGattBackend, self).__init__(address, async, debug)
  File "/home/pymetawear/pymetawear/backends/__init__.py", line 53, in __init__
    METAWEAR_SERVICE_NOTIFY_CHAR[1])
  File "/home/pymetawear/pymetawear/backends/pygatt/__init__.py", line 104, in get_handle
    return self.requester.get_handle(uuid) + int(notify_handle)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 192, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: <BLEError No characteristic found matching 326a9006-85cb-9195-d9dd-464cfbbae75a>
edison@ubilinux:/home/pymetawear/examples$
`

But characteristic "326a9006-85cb-9195-d9dd-464cfbbae75a" can be found by sending "characteristics" command in gatttool.  See last characteristic matches.

```
edison@ubilinux:/home/pymetawear/examples$ sudo gatttool -I
[                 ][LE]> connect CA:CD:4F:09:A8:DE random
Attempting to connect to CA:CD:4F:09:A8:DE
Connection successful
[CA:CD:4F:09:A8:DE][LE]> characteristics
handle: 0x0002, char properties: 0x0a, char value handle: 0x0003, uuid: 00002a00-0000-1000-8000-00805f9b34fb
handle: 0x0004, char properties: 0x02, char value handle: 0x0005, uuid: 00002a01-0000-1000-8000-00805f9b34fb
handle: 0x0006, char properties: 0x02, char value handle: 0x0007, uuid: 00002a04-0000-1000-8000-00805f9b34fb
handle: 0x0009, char properties: 0x20, char value handle: 0x000a, uuid: 00002a05-0000-1000-8000-00805f9b34fb
handle: 0x000d, char properties: 0x12, char value handle: 0x000e, uuid: 00002a19-0000-1000-8000-00805f9b34fb
handle: 0x0011, char properties: 0x02, char value handle: 0x0012, uuid: 00002a29-0000-1000-8000-00805f9b34fb
handle: 0x0013, char properties: 0x02, char value handle: 0x0014, uuid: 00002a24-0000-1000-8000-00805f9b34fb
handle: 0x0015, char properties: 0x02, char value handle: 0x0016, uuid: 00002a25-0000-1000-8000-00805f9b34fb
handle: 0x0017, char properties: 0x02, char value handle: 0x0018, uuid: 00002a27-0000-1000-8000-00805f9b34fb
handle: 0x0019, char properties: 0x02, char value handle: 0x001a, uuid: 00002a26-0000-1000-8000-00805f9b34fb
handle: 0x001c, char properties: 0x0e, char value handle: 0x001d, uuid: 326a9001-85cb-9195-d9dd-464cfbbae75a
handle: 0x001e, char properties: 0x12, char value handle: 0x001f, uuid: 326a9006-85cb-9195-d9dd-464cfbbae75a
[CA:CD:4F:09:A8:DE][LE]>

```

hbldh, 2016-04-19:
I have explored somewhat on my Raspberry Pi, which had 4.99 and not getting anything to work.
I do not know what to do to get it to work on that version, and I feel it is an unnecessary path to explore. 

I would explore the possibility to use a distribution based on the ``jessie`` version of Debian instead of ``wheezy``:

http://www.robinkirkman.com/jubilinux/
https://github.com/catmaker/chippy/wiki/Upgrading-Ubilinux-Debian-Wheezy-to-Jessie

siddrajput, 2016-04-23:
I updated to jessie but I'm still getting the same error:

hbldh, 2016-04-27:
Found two bugs that might alleviate your problems (one regarding `pybluez` timeout and one regarding switch notifications); both of which are fixed in the new `0.4.3` release. Clean away any old `pymetawear` code and install anew from the `master` branch:
```
pip install git+git://github.com/hbldh/pymetawear.git
```

siddrajput, 2016-04-28:
With 'pybluez' I'm still getting:

`edison@ubilinux:~/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTRequester...
Connecting GATTRequester...
Subscribe 0x001f:
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/edison/pymetawear/pymetawear/backends/__init__.py", line 109, in mbl_mw_read_gatt_char
    response = self.read_gatt_char_by_uuid(characteristic_uuid)
  File "/home/edison/pymetawear/pymetawear/backends/pybluez/__init__.py", line 112, in read_gatt_char_by_uuid
    return self.requester.read_by_uuid(str(characteristic_uuid))[0]
RuntimeError: Characteristic value/descriptor operation failed: No attribute found within the given range
Waiting for MetaWear board to be fully initialized...
`

And with 'pygatt' I'm getting:

`edison@ubilinux:~/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTToolBackend and starting GATTtool process...
Connecting GATTTool...
Subscribe 0x001f:
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/home/edison/pymetawear/pymetawear/backends/__init__.py", line 109, in mbl_mw_read_gatt_char
    response = self.read_gatt_char_by_uuid(characteristic_uuid)
  File "/home/edison/pymetawear/pymetawear/backends/pygatt/__init__.py", line 91, in read_gatt_char_by_uuid
    return self.requester.char_read(str(characteristic_uuid))
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/device.py", line 15, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/device.py", line 38, in char_read
    return self._backend.char_read(self, uuid, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 37, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 352, in char_read
    self._expect('value: .*? \r')
  File "/usr/local/lib/python2.7/dist-packages/pygatt/backends/gatttool/gatttool.py", line 302, in _expect
    "Timed out waiting for a notification")
pygatt.exceptions.NotificationTimeout: <NotificationTimeout Timed out waiting for a notification>
Waiting for MetaWear board to be fully initialized...
`

siddrajput, 2016-04-28:
I'm also getting this still:

`edison@ubilinux:~/pymetawear/examples$ sudo python led.py
Discovering nearby MetaWear boards...
Creating new GATTToolBackend and starting GATTtool process...
Connecting GATTTool...
No handlers could be found for logger "pygatt.backends.gatttool.gatttool"
Traceback (most recent call last):
  File "led.py", line 28, in <module>
    c = MetaWearClient(str(address), 'pygatt', debug=True)
  File "/home/edison/pymetawear/pymetawear/client.py", line 104, in __init__
    self._address, timeout=timeout, debug=debug)
  File "/home/edison/pymetawear/pymetawear/backends/pygatt/__init__.py", line 39, in __init__
    debug)
  File "/home/edison/pymetawear/pymetawear/backends/__init__.py", line 56, in __init__
    METAWEAR_SERVICE_NOTIFY_CHAR[1])
  File "/home/edison/pymetawear/pymetawear/backends/pygatt/__init__.py", line 112, in get_handle
    return self.requester.get_handle(uuid) + int(notify_handle)
  File "/home/edison/pymetawear/pymetawear/backends/pygatt/gatttool.py", line 48, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: <BLEError No characteristic found matching 326a9006-85cb-9195-d9dd-464cfbbae75a>
`

brianhouse, 2016-08-07:
hi guys -- I think I'm in a similar situation. I've got a Raspi B 3, running Jessie, and I pulled the latest master from pymetawear.

With pygatt I'm getting this:

pi@raspberrypi:~ $ sudo python accelerometer.py
Discovering nearby MetaWear boards...
Creating new GATTToolBackend and starting GATTtool process...
Connecting GATTTool...
Subscribe 0x001f: 
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.4.4-py2.7.egg/pymetawear/backends/__init__.py", line 109, in mbl_mw_read_gatt_char
    response = self.read_gatt_char_by_uuid(characteristic_uuid)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.4.4-py2.7.egg/pymetawear/backends/pygatt/__init__.py", line 91, in read_gatt_char_by_uuid
    return self.requester.char_read(str(characteristic_uuid))
  File "/usr/local/lib/python2.7/dist-packages/pygatt-2.0.1-py2.7.egg/pygatt/backends/gatttool/device.py", line 15, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt-2.0.1-py2.7.egg/pygatt/backends/gatttool/device.py", line 38, in char_read
    return self._backend.char_read(self, uuid, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt-2.0.1-py2.7.egg/pygatt/backends/gatttool/gatttool.py", line 37, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pygatt-2.0.1-py2.7.egg/pygatt/backends/gatttool/gatttool.py", line 352, in char_read
    self._expect('value: .*? \r')
  File "/usr/local/lib/python2.7/dist-packages/pygatt-2.0.1-py2.7.egg/pygatt/backends/gatttool/gatttool.py", line 302, in _expect
    "Timed out waiting for a notification")
pygatt.exceptions.NotificationTimeout: <NotificationTimeout Timed out waiting for a notification>
Waiting for MetaWear board to be fully initialized...

And with pybluez:

pi@raspberrypi:~ $ sudo python accelerometer.py 
Discovering nearby MetaWear boards...
Creating new GATTRequester...
Connecting GATTRequester...
Subscribe 0x001f: 
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 314, in 'calling callback function'
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.4.4-py2.7.egg/pymetawear/backends/__init__.py", line 109, in mbl_mw_read_gatt_char
    response = self.read_gatt_char_by_uuid(characteristic_uuid)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.4.4-py2.7.egg/pymetawear/backends/pybluez/__init__.py", line 112, in read_gatt_char_by_uuid
    return self.requester.read_by_uuid(str(characteristic_uuid))[0]
RuntimeError: Characteristic value/descriptor operation failed: No attribute found within the given range
Waiting for MetaWear board to be fully initialized...

I'm able to connect with gatttool without a problem.

Any further thoughts? Excited about this module, would love to have it running on the pi.

hbldh, 2016-08-08:
@brianhouse I have a Raspberry Pi 2 running the Jessie image as well, and I have experienced this problem too. I am not sure why right now, and I have not been able to find time to explore it further yet. I think it has to do with the fact that the low energy components of pybluez is using the pygattlib library and it bundles an old version of bluez (https://bitbucket.org/OscarAcena/pygattlib/issues/42/raspberry-pi-discovery-and-primary-service). 
As to the pygatt not working I am unsure as to why. Debugging it requires some pdb action during connection to find out why...

hbldh, 2016-08-10:
@brianhouse 
I believe the problem is within the building of MetaWear-CppAPI on Raspberry. I have tried building it on my Raspberry 2, but cannot get it to work properly. 
Did you get PyMetaWear to install properly on your Raspberry 3? Did you install it with pip?

I have logged an [issue](mbientlab/Metawear-CppAPI#7) on the C++ API repo to see if it can be solved with some help from them.

brianhouse, 2016-08-10:
@hbldh 
I cloned the pymetawear repo, and built it with setup.py. Out of the box, that actually fails, because the Pi 3 didnt like the "-m32" flag in the MetaWear-CppAPI Makefile. So I edited the Makefile to comment out the architecture flags and built again, which succeeded. I did that with both python 2.7 and 3.4, which end up with similar runtime errors, though the log output is a bit more convoluted with 3.4, maybe there's some additional threading going on. I've not tried to do anything with MetaWear-CppAPI on its own.

Hope that helps, let me know if you want me to try anything.

hbldh, 2016-08-10:
I also did that, and then tried running the tests on the MetaWear-CppAPI shared library file, and no tests can be run whatsoever (see the issue referenced above for some more info on that) using the built file.

Good to know that you had the same problems as me though; that's a start at least :)

hbldh, 2016-08-16:
@brianhouse With the modifications done in MetaWear-CppAPI and these being integrated into PyMetaWear, I can now `pip install -e .` and use it on my Raspberry Pi 2, both with `pybluez` and `pygatt` backends.
The `develop` branch of PyMetaWear is now using MetaWear-CppAPI version 0.5.0, and this will be merged eventually when some further modifications have been done.

@siddrajput You might want to try to install the PyMetaWear develop branch on your Edison once again and see if it works now. The problem of your platform might be related to the Raspberry Pi case.

brianhouse, 2016-08-16:
@hbldh
- installs smoothly!!
- led.py runs as expected
- battery.py runs as expected
- accelerometer.py runs without an error -- but I dont receive any signal notifications. I've verified that the MetaWear is working with an iOS app, and I've tried lowering the data rate, but nothing is coming through. I'm also able to print the accelerometer object and see `Accelerometer Mma8452q: Data rates (Hz): [1.56, 6.25, 12.5, 50.0, 100.0, 200.0, 400.0, 800.0], Data ranges (g): [2.0, 4.0, 8.0]`

hbldh, 2016-08-17:
Good, then we are one step closer to functionality at least...

I have not tried a board with the `Mma8452q` sensor, but it should work as expected. You do own a MetaWear R board, right? Just checking so that PyMetaWear has not made an error in detecting which accelerometer your board has...

I have a MetaWear R Pro, and it did not give correct output in accelerometer example either when I tried, neither from desktop, RPi or iOS app. I did a soft reset (either from the MetaWear iOS app or by running
```
from pymetawear.client import MetaWearClient
address = 'FF:50:35:82:3B:5A'
c = MetaWearClient(str(address), 'pygatt', debug=True)
c.soft_reset()
c.disconnect()
```
) Then try running the accelerometer script again.

brianhouse, 2016-08-19:
I have a few MetaWear Rs (with the Mma8452q) and one MetaWear C (with the BMI160).

As it turn out, the C works just fine -- I get readings right away. But none of the Rs respond. I don't get why the accelerometer would make a difference, but it looks like it might.

I just ordered an RG, so I can verify whether it's something else related to the series.

brianhouse, 2016-08-23:
Confirmed that RGs work.

hbldh, 2016-08-26:
The RGs have Bosch accelerometers as well as the Cs. It seems to be a problem exclusive to the MMA8452q sensor then.

I created a [Gist with two python scripts](https://gist.github.com/hbldh/d1c8513eb15611a488f6b6c423be2839). They both use the `libmetawear` shared library in its raw form, only using the MetaWearClient for connection. Try especially the `accelerometer_mma8452q.py` file to see if it works calling the specific sensor methods instead of the generic methods (as is done in `accelerometer_generic.py` in the Gist).

brianhouse, 2016-08-30:
With both scripts, I dont see callbacks for the R, though I am seeing Notify statements stream by.

Will it be possible to use the high frequency functionality?
https://mbientlab.com/cppdocs/latest/accelerometer.html#high-frequency-stream

hbldh, 2016-09-06:
So the notification printouts show up like this:
```
...
Notify 0x001f: 03 04 04 ff 7b 01 21 10
Notify 0x001f: 03 04 14 ff 84 01 21 10
Notify 0x001f: 03 04 0c ff 83 01 2d 10
Notify 0x001f: 03 04 1a ff 88 01 29 10
Notify 0x001f: 03 04 0c ff 88 01 24 10
...
``` 
?
But the callback function you specified is not called?

The high frequency methods can definitely be used. Switch `mbl_mw_acc_mma8452q_get_acceleration_data_signal` to `mbl_mw_acc_mma8452q_get_high_freq_acceleration_data_signal` and try again. However, if it is due to callbacks not being registered properly, then it will probably not help.

hbldh, 2016-09-14:
@brianhouse High frequency functionality is now present in `master` branch: see [examples/accelerometer.py](https://github.com/hbldh/pymetawear/blob/master/examples/accelerometer.py#L35). Not sure if it will help with the problems with the R series, but just so you know.

Have you tested anything more? Will close this issue in a few days if nothing more surfaces.

brianhouse, 2016-09-19:
thanks @hbldh. Ive been on another project and havent been able to test -- excited to test high frequency. but I think this issue can be closed regardless.

closed 2016-09-20 by hbldh