hbldh/pymetawear #15: Issues with other modules?

fraguada, 2016-09-03:
I have two scrips that work fine independently.  Once uses OpenCV and one uses PyMetawear.  When I try to combine them, things stop working.  
Here is something similar to what I'm doing in open cv: https://gist.github.com/fraguada/42eeba928cf85b8ecc989b74d69d3635
Here is something similar to what I'm doing with PyMetawear: https://gist.github.com/fraguada/71456b09cf585c57206a52814e93c2ce

Both of these work independently.  Now when I try to do something like this: https://gist.github.com/fraguada/e773518448ed1b7a6e34ea7cee0a68da (just importing the pymetawear modules) the script stops working.  I do not get any errors, the script doesn't open the opencv window.  If I get out of the script with keyboard interrupt, the camera is unavailable on the next run.

I'm on a Raspberry Pi 3 running Raspbian Debian Jessie and using Python 3.4.2 and I'm obviously not well versed in python, its modules, etc.  Any pointers on how I could debug this would be appreciated.

hbldh, 2016-09-05:
Hm, this was strange. And the scripts stop working just by importing the pymetawear module?
Could you try to run it using Python 2.7 as well and see if that changes anything, provided that you built opencv with python2 support as well? 

I will check it out sometime later during this week; I am preoccupied until Wednesday.

fraguada, 2016-09-06:
> And the scripts stop working just by importing the pymetawear module?
That is correct.  Through I have not been able to trace the actual cause.

I have not built OpenCV for python 2, but I can give it a shot to see if it will make a difference.

hbldh, 2016-09-08:
I have now [built and installed opencv](http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/) for both Python 2 and Python 3 on my Raspberry Pi 2, with a mounted camera.

In both versions, the OpenCV windows opens (nothing is display in it though) and I regretfully have to tell you that I cannot replicate your issue. No difference between having the pymetawear imported or not could be observed.

I installed in virtualenvs, symlinking OpenCV into it. How does this correlate to your installation?

closed 2016-10-13 by hbldh