hbldh/pymetawear #18: pygatt not working

enlight3d, 2016-09-14:
Hello, I was using your previous version since last week (i think it was 0.4.4) but as I've since your latest commits, I wanted to update to the latest working version. I tried master, develop and then figured out that I maybe needed to try pygatt-py3 which is the version that I'm on now. Only problem is that even when loading your examples (accelerometer for example) well I have this error (see image).

[image]

I'm on a raspberry pi 3 with latest jessie version.

hbldh, 2016-09-14:
The `pygatt-py3` branch is a preemptive branch, preparing for predicted changes in the next `pygatt` release (it cannot be used right now except with the `python-3-tests` branch of `pygatt`), so avoid that one.

The problem you encountered is however closely related to that. Apparently I had mistakenly removed an overriding method that monkey-patched that `basestring` error in `pygatt` to early. I have replaced it now in both develop and master branch (version 0.5.0a3) and tested it on my RPi 2 with Jessie and Python 3.4.

closed 2016-09-14 by hbldh

enlight3d, 2016-09-14:
okay thanks !

commit 3deb5425 (2016-09-14): Critical fix for Python 3 use of pygatt backend.

Fixes #18