hbldh/pymetawear #20: No bluetooth device found on Ubuntu bash (this is a new feature in Wndows 10)

livi0, 2016-10-13:
Hi, 

Is there any solution please to be able to activate the Bluetooth device from the Ubuntu bash? 
hcitool dev
gives no device

By running your example led.py from Spyder I get this error:

>>> runfile('/mnt/c/Users/iLivi/Documents/Linux/pymetawear/examples/led.py', wdir=r'/mnt/c/Users/iLivi/Documents/Linux/pymetawear/examples')
Discovering nearby Bluetooth Low Energy devices...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/spyderlib/widgets/externalshell/sitecustomize.py", line 540, in runfile
    execfile(filename, namespace)
  File "/mnt/c/Users/iLivi/Documents/Linux/pymetawear/examples/led.py", line 22, in <module>
    address = scan_and_select_le_device()
  File "discover.py", line 26, in scan_and_select_le_device
    ble_devices = discover_devices(timeout=timeout)
  File "/mnt/c/Users/iLivi/Documents/Linux/pymetawear/pymetawear/client.py", line 70, in discover_devices
    os.kill(p.pid, signal.SIGINT)
OSError: [Errno 3] No such process

May you please help out? Thanks!

hbldh, 2016-10-13:
I am not sure if you can get access to Bluetooth from the Ubuntu Bash in Windows 10, but it would be nice to see if it is possible! First off, you need to install BlueZ:

```
sudo apt-get install bluez
```
That will install the Linux Bluetooth library, including the `hcitool` application which was the one leading to the error above. A problem might arise if you get Bluez 4.X installed from the Ubuntu repositories ([pygatt](https://github.com/peplin/pygatt) which I use underneath the hood requires 5.X), but lets try and see.

livi0, 2016-10-13:
It seems that bluez is already installed, probably I already installed the "Additional requirements for pybluez". Here it's what I get by "sudo apt-get install bluez"

liva@DESKTOP-119GS08:~$ sudo apt-get install bluez
Reading package lists... Done
Building dependency tree
Reading state information... Done
bluez is already the newest version.
bluez set to manually installed.
The following package was automatically installed and is no longer required:
  os-prober
Use 'apt-get autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
liva@DESKTOP-119GS08:~$ hcitool dev
Devices:
liva@DESKTOP-119GS08:~$

livi0, 2016-10-13:
BTW, when doing "sudo pip install pymetawear", there is a warning of a syntax error. See hereafter. I don't know if it's related to the current issue.

....
    no previously-included directories found matching 'pymetawear/Metawear-CppAPI/.git'
    no previously-included directories found matching 'docs/build'
    warning: no previously-included files matching '__pycache__/*' found anywhere in distribution
  Running setup.py install for pygatt

    warning: no files found matching 'NOTICE'
Compiling /tmp/pip_build_root/pexpect/pexpect/async.py ...
  File "/tmp/pip_build_root/pexpect/pexpect/async.py", line 16
    transport, pw = yield from asyncio.get_event_loop()\
                             ^
SyntaxError: invalid syntax

Successfully installed pymetawear pygatt pexpect enum34 ptyprocess
Cleaning up...
liva@DESKTOP-119GS08:~$

livi0, 2016-10-13:
I tried to use the Bash on Ubuntu on Windows because I'd need a program to read MetaWear sensors and communicate the data further to another Windows based program. I've seen some C#, C++ libraries for MetaWear, but I don't know those languages (only some old C). I usual use Matlab, so I thought this Python library would be easier (Python has some Matlab similitudes) to make it work and make an executable in Windows. Since this library is for Linux only, I thought that the Ubuntu bash would be the way to go. However, if this library could run on Anaconda Python in Windows, that would be way more promising. Any hope to get a Windows version of this pymetawear library? Thanks!

hbldh, 2016-10-13:
The Syntax error during install is not the real problem, but something I will look into more closely.

The 
```
liva@DESKTOP-119GS08:~$ hcitool dev
Devices:
```
however seems to indicate that Ubuntu Bash in Windows 10 does not get access to the Bluetooth. hardware. I have not tried it myself yet, so I cannot help you with it I'm afraid. Send some questions to Microsoft support and maybe they'll have a good answer.

Apart from that, PyMetaWear builds on Windows (given that Visual Studio 2015 Community is installed); the problem is the Python BLE communication packages for Windows. Very few have Windows support. Pygatt (the one I'm using) has BLE in WIndows, but only for a certain range of Bluetooth USB Dongles.

I have some ideas about how it might be able to be done in Windows 10, but no time to explore it right now I'm afraid.

livi0, 2016-10-13:
Now trying on Anaconda under Win10.
pybluez and bluepy both fail during installation! See hereafter!
Liviu

C:\WINDOWS\system32>pip install pymetawear[pybluez]
Requirement already satisfied (use --upgrade to upgrade): pymetawear[pybluez] in c:\users\ilivi\anaconda3\lib\site-packages
Requirement already satisfied (use --upgrade to upgrade): pygatt[gatttool]>=3.0.0 in c:\users\ilivi\anaconda3\lib\site-packages (from pymetawear[pybluez])
Requirement already satisfied (use --upgrade to upgrade): pexpect>=4.2.0 in c:\users\ilivi\anaconda3\lib\site-packages (from pymetawear[pybluez])
Collecting pybluez[ble]>=0.22; extra == "pybluez" (from pymetawear[pybluez])
  Downloading PyBluez-0.22.zip (109kB)
    100% |################################| 112kB 255kB/s
Requirement already satisfied (use --upgrade to upgrade): enum34 in c:\users\ilivi\anaconda3\lib\site-packages (from pygatt[gatttool]>=3.0.0->pymetawear[pybluez])
Requirement already satisfied (use --upgrade to upgrade): pyserial in c:\users\ilivi\anaconda3\lib\site-packages (from pygatt[gatttool]>=3.0.0->pymetawear[pybluez])
Requirement already satisfied (use --upgrade to upgrade): ptyprocess>=0.5 in c:\users\ilivi\anaconda3\lib\site-packages (from pexpect>=4.2.0->pymetawear[pybluez])
Collecting gattlib==0.20150805 (from pybluez[ble]>=0.22; extra == "pybluez"->pymetawear[pybluez])
  Downloading gattlib-0.20150805.tar.gz (1.7MB)
    100% |################################| 1.7MB 474kB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\iLivi\AppData\Local\Temp\pip-build-605r7rr5\gattlib\setup.py", line 48, in <module>
        raise OSError("Not supported OS")
    OSError: Not supported OS

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\iLivi\AppData\Local\Temp\pip-build-605r7rr5\gattlib\

C:\WINDOWS\system32>pip install pymetawear[bluepy]
Requirement already satisfied (use --upgrade to upgrade): pymetawear[bluepy] in c:\users\ilivi\anaconda3\lib\site-packages
Requirement already satisfied (use --upgrade to upgrade): pygatt[gatttool]>=3.0.0 in c:\users\ilivi\anaconda3\lib\site-packages (from pymetawear[bluepy])
Requirement already satisfied (use --upgrade to upgrade): pexpect>=4.2.0 in c:\users\ilivi\anaconda3\lib\site-packages (from pymetawear[bluepy])
Collecting bluepy>=1.0.5; extra == "bluepy" (from pymetawear[bluepy])
  Downloading bluepy-1.0.5.tar.gz (210kB)
    100% |################################| 215kB 389kB/s
Requirement already satisfied (use --upgrade to upgrade): pyserial in c:\users\ilivi\anaconda3\lib\site-packages (from pygatt[gatttool]>=3.0.0->pymetawear[bluepy])
Requirement already satisfied (use --upgrade to upgrade): enum34 in c:\users\ilivi\anaconda3\lib\site-packages (from pygatt[gatttool]>=3.0.0->pymetawear[bluepy])
Requirement already satisfied (use --upgrade to upgrade): ptyprocess>=0.5 in c:\users\ilivi\anaconda3\lib\site-packages (from pexpect>=4.2.0->pymetawear[bluepy])
Building wheels for collected packages: bluepy
  Running setup.py bdist_wheel for bluepy ... error
  Complete output from command C:\Users\iLivi\Anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\iLivi\\AppData\\Local\\Temp\\pip-build-2ep7xmq2\\bluepy\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d C:\Users\iLivi\AppData\Local\Temp\tmp34xixmvhpip-wheel- --python-tag cp35:
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib
  creating build\lib\bluepy
  copying bluepy\blescan.py -> build\lib\bluepy
  copying bluepy\btle.py -> build\lib\bluepy
  copying bluepy\get_services.py -> build\lib\bluepy
  copying bluepy\sensortag.py -> build\lib\bluepy
  copying bluepy\__init__.py -> build\lib\bluepy
  copying bluepy\uuids.json -> build\lib\bluepy
  copying bluepy\bluez-src.tgz -> build\lib\bluepy
  copying bluepy\bluepy-helper.c -> build\lib\bluepy
  copying bluepy\Makefile -> build\lib\bluepy
  installing to build\bdist.win-amd64\wheel
  running install
  Working dir is C:\Users\iLivi\AppData\Local\Temp\pip-build-2ep7xmq2\bluepy
  execute make -C ./bluepy clean
  error: [WinError 2] The system cannot find the file specified

  ----------------------------------------
  Failed building wheel for bluepy
  Running setup.py clean for bluepy
Failed to build bluepy
Installing collected packages: bluepy
  Running setup.py install for bluepy ... error
    Complete output from command C:\Users\iLivi\Anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\iLivi\\AppData\\Local\\Temp\\pip-build-2ep7xmq2\\bluepy\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\iLivi\AppData\Local\Temp\pip-sqdh84fk-record\install-record.txt --single-version-externally-managed --compile:
    running install
    Working dir is C:\Users\iLivi\AppData\Local\Temp\pip-build-2ep7xmq2\bluepy
    execute make -C ./bluepy clean
    error: [WinError 2] The system cannot find the file specified

    ----------------------------------------
Command "C:\Users\iLivi\Anaconda3\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\iLivi\\AppData\\Local\\Temp\\pip-build-2ep7xmq2\\bluepy\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record C:\Users\iLivi\AppData\Local\Temp\pip-sqdh84fk-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\iLivi\AppData\Local\Temp\pip-build-2ep7xmq2\bluepy\

C:\WINDOWS\system32>

livi0, 2016-10-13:
By trying to use  pygatt, like this

from pymetawear.client import MetaWearClient
backend = 'pygatt'  # Or 'pybluez' or 'bluepy'
c = MetaWearClient('45:FD:49:78:0A:02', backend)

I got an error of missing module "termios", which is not a Windows module. Should this mean that pygatt cannot be used on Windows? Hereafter is the detailed error:

 "pygatt[GATTTOOL] package error :{0}".format(PyGattBackend))
pymetawear.exceptions.PyMetaWearException: pygatt[GATTTOOL] package error :No module named 'termios'
Traceback (most recent call last):

hbldh, 2016-10-17:
Installed in Ubuntu Bash on Windows and, as I expected, it will not work due Bluetooth hardware not being available in the Ubuntu Sandbox. I did not experience the `termios` problem.
You will have to install it into a proper Linux system, or find/write a Python library that does BLE communication in Windows, sorry.

closed 2016-10-19 by hbldh