hbldh/pymetawear #29: C.H.I.P computer freeze using pymetawear

gilduz, 2017-03-14:
Hello everyone!

I'm using pymetawear on a raspberry pi model B without any problem. Everything works good and the stability is perfect.
If I try to move my project on a CHIP computer ( https://getchip.com ) the device completely freeze after acquiring a bunch of accelerometer samples. The only way I have to interact with the shell again is to turn off CHIP computer (unplugging the power adapter) an turn on it again.
Someone noticed the same problematics?
The same problem can be reproduced with accelerometer.py example setting 1hz as samplerate and incrementing the delay to wait for a freeze.
Both devices (raspberry and CHIP) have jessie installed. Not sure if the problem is related to pymetawear or C++ metawear library, or CHIP computer itself.

hbldh, 2017-03-14:
Interesting. I've got a CHIP myself, but haven't tried it with pymetawear. Haven't got time this week, but maybe you can answer these questions first:

- [ ] Does it freeze both with and without debug turned on? 

- [ ] If you modify `accelerometer.py` so that it does not print all received data, but stores it in an array instead, does it still freeze on the same sample rate setting?

- [ ] Does it freeze using both `pygatt` and `pybluez` backends?

gilduz, 2017-03-14:
- I tried to disabling debug without any improvements
- All test was done with pygatt, I will try later with pybluez. If I remember well I tried to install pybluez but some compilation error occurred

I'll be able to reply better to your question in some hours... Just the time to come back home, reflash CHIP (after a lot of forced shutdown the image now is corrupted) and try again. Nice you have the same board to reproduce the error!

hbldh, 2017-03-15:
The `pybluez` backend might be tricky to get to work on some systems. I will try it out eventually...

gilduz, 2017-03-15:
I had time to do some investigation. I reflash my CHIP with headless 4.4 image and I tried pymetawear with pygatt (pip install pymetawear[pybluez] failed during "building wheel for gattlib")
With pygatt, the problem is different than described before. 
I'm able to use pymetawear if I connect to CHIP computer via serial USB interface. If I use ssh to run the modified accelerometer.py (1 hz, 200 secs delay), after some seconds my ssh shell freeze, and the CHIP wifi connection became messed up (I reported the problem also here [https://bbs.nextthing.co/t/chip-becomes-unreachable-over-ssh/12467/15](https://bbs.nextthing.co/t/chip-becomes-unreachable-over-ssh/12467/15) )  
So, not all CHIP freeze, but just wifi connection, after the problem, I'm still able to get a shell using serial USB.
Probably is a CHIP problematic, not really sure if the combination of WiFi and BLE is the cause.
Note: Power management is turned off following these instructions [https://github.com/fordsfords/wlan_pwr/tree/gh-pages](https://github.com/fordsfords/wlan_pwr/tree/gh-pages)

hbldh, 2017-04-17:
Sorry, I have not been able to find time to do any tests on this, and I don't think I will be able to within a foreseeable future... Have you had any luck with it or just left it?

pybluez uses pygattlib (https://bitbucket.org/OscarAcena/pygattlib) and it is hardcoded to link to boost compiled for python 3.4, which fails on e.g. Jessie that has Python 3.5 instead.

Workaround: Clone repo, make changes according to [this PR](https://bitbucket.org/OscarAcena/pygattlib/pull-requests/18/setuppy-boost_python-library-linking-fix/diff), install pygattlib from that and then install pybluez after that and finally install pymetawear.

closed 2017-07-07 by mbientlab