hbldh/pymetawear #30: pymetawear + Raspberry Pi 3

mdgagne, 2017-04-01:
I'm having some problems connecting to a metawear sensor from a raspberry pi 3 using pymetawear. I'm hoping someone could provide feedback on their experience with this library on raspberry pi3 + jessie. Here is the test app I'm executing
```
from pymetawear.client import MetaWearClient
backend = 'pygatt'  # Or 'pybluez'
c = MetaWearClient('E9:D1:C1:68:0D:86', backend, 'hci0', None, True, True)

pattern = c.led.load_preset_pattern('blink', repeat_count=10)
c.led.write_pattern(pattern, 'g')
c.led.play()
```
And the console log output. Sometimes it also fails with a plain old timeout.
```
2017-04-01 13:13:21,012 DEBUG Added a <stdout> logging handler to logger: pymetawear
2017-04-01 13:13:21,013 INFO Creating MetaWearClient for E9:D1:C1:68:0D:86...
2017-04-01 13:13:21,013 INFO PyGattBackend: Creating new GATTToolBackend and starting GATTtool process...
2017-04-01 13:13:21,063 INFO PyGattBackend: Connecting to E9:D1:C1:68:0D:86 using GATTTool...
No handlers could be found for logger "pygatt.device"
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    c = MetaWearClient('E9:D1:C1:68:0D:86', backend, 'hci0', None, True, True)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear/client.py", line 95, in __init__
    self.connect()
  File "/usr/local/lib/python2.7/dist-packages/pymetawear/client.py", line 128, in connect
    self.backend.connect(clean_connect=clean_connect)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear/backends/pygatt.py", line 96, in connect
    super(PyGattBackend, self).connect()
  File "/usr/local/lib/python2.7/dist-packages/pymetawear/backends/__init__.py", line 96, in connect
    METAWEAR_SERVICE_NOTIFY_CHAR[1])
  File "/usr/local/lib/python2.7/dist-packages/pymetawear/backends/pygatt.py", line 145, in get_handle
    return self.requester.get_handle(characteristic_uuid) + int(notify_handle)
  File "/usr/local/lib/python2.7/dist-packages/pygatt/device.py", line 195, in get_handle
    raise exceptions.BLEError(message)
pygatt.exceptions.BLEError: No characteristic found matching 326a9006-85cb-9195-d9dd-464cfbbae75a
```

hbldh, 2017-04-17:
Are you sure that the ``E9:D1:C1:68:0D:86`` address is really a MetaWear board?
If it is, what firmware version is your MetaWear board running?

mdgagne, 2017-04-22:
Yes. Here is the info from MetaWear Android app.

Model number 0
Serial 342a
FW 1.2.5
HD rev 0.2

closed 2017-07-07 by mbientlab