hbldh/pymetawear #39: No output from accelerometer example

somacoder, 2017-07-25:
I am able to successfully connect to my MetawearRG using my Raspberry Pi Zero, and subsequently blink the led using the led script. However, when running the accelerometer example, I am not seeing any output from the notification callback function. Even when running in debug mode, nothing happens after running the c.accelerometer.notifications(acc_callback) line. Is this a known issue, or am I missing something here? Thanks!

scaryghost, 2017-07-27:
* Does the board work with the Android or iOS apps?  
* Which btle backend are you using?  
* What are the contents of the debug output?

somacoder, 2017-07-31:
-Board works with Android and iOS
-Latest github release (as of 7-31-17)
-pygatt
-Debug Output (From examples/accelerometer.py):

2017-07-31 12:14:11,822 DEBUG Added a <stdout> logging handler to logger: pymetawear
2017-07-31 12:14:11,825 INFO Creating MetaWearClient for FC:3A:98:16:8A:0F...
2017-07-31 12:14:11,828 INFO PyGattBackend: Creating new GATTToolBackend and starting GATTtool process...
2017-07-31 12:14:11,831 INFO Backend starter with pygatt for device address FC:3A:98:16:8A:0F with timeout None...
2017-07-31 12:14:11,930 INFO PyGattBackend: Connecting to FC:3A:98:16:8A:0F using GATTTool...
2017-07-31 12:14:15,080 DEBUG Subscribe 0x001f: 
2017-07-31 12:14:15,487 DEBUG Write  0x001d: 01 80
2017-07-31 12:14:15,550 DEBUG Read   0x0014: 31
2017-07-31 12:14:15,554 DEBUG Read   0x001a: 31 2e 32 2e 35
2017-07-31 12:14:15,556 DEBUG Waiting for MetaWear board to be fully initialized...
2017-07-31 12:14:15,672 DEBUG Notify 0x001f: 01 80 00 00
2017-07-31 12:14:15,676 DEBUG Write  0x001d: 02 80
2017-07-31 12:14:15,875 DEBUG Notify 0x001f: 02 80 00 01 03 00
2017-07-31 12:14:15,880 DEBUG Write  0x001d: 03 80
2017-07-31 12:14:16,078 DEBUG Notify 0x001f: 03 80 01 01
2017-07-31 12:14:16,083 DEBUG Write  0x001d: 04 80
2017-07-31 12:14:16,285 DEBUG Notify 0x001f: 04 80 01 00 00 03 01 02
2017-07-31 12:14:16,290 DEBUG Write  0x001d: 05 80
2017-07-31 12:14:16,484 DEBUG Notify 0x001f: 05 80 00 02 03 03 03 03 01 01 01 01
2017-07-31 12:14:16,490 DEBUG Write  0x001d: 06 80
2017-07-31 12:14:16,685 DEBUG Notify 0x001f: 06 80 00 00
2017-07-31 12:14:16,690 DEBUG Write  0x001d: 07 80
2017-07-31 12:14:16,888 DEBUG Notify 0x001f: 07 80 00 00
2017-07-31 12:14:16,893 DEBUG Write  0x001d: 08 80
2017-07-31 12:14:17,094 DEBUG Notify 0x001f: 08 80 00 00
2017-07-31 12:14:17,098 DEBUG Write  0x001d: 09 80
2017-07-31 12:14:17,293 DEBUG Notify 0x001f: 09 80 00 00 1c
2017-07-31 12:14:17,298 DEBUG Write  0x001d: 0a 80
2017-07-31 12:14:17,496 DEBUG Notify 0x001f: 0a 80 00 00 1c
2017-07-31 12:14:17,500 DEBUG Write  0x001d: 0b 80
2017-07-31 12:14:17,699 DEBUG Notify 0x001f: 0b 80 00 02 08 80 1f 00 00
2017-07-31 12:14:17,704 DEBUG Write  0x001d: 0c 80
2017-07-31 12:14:17,901 DEBUG Notify 0x001f: 0c 80 00 00 08
2017-07-31 12:14:17,905 DEBUG Write  0x001d: 0d 80
2017-07-31 12:14:18,103 DEBUG Notify 0x001f: 0d 80 00 01
2017-07-31 12:14:18,108 DEBUG Write  0x001d: 0f 80
2017-07-31 12:14:18,306 DEBUG Notify 0x001f: 0f 80 00 01 08
2017-07-31 12:14:18,310 DEBUG Write  0x001d: 10 80
2017-07-31 12:14:18,508 DEBUG Notify 0x001f: 10 80
2017-07-31 12:14:18,514 DEBUG Write  0x001d: 11 80
2017-07-31 12:14:18,714 DEBUG Notify 0x001f: 11 80 00 04
2017-07-31 12:14:18,718 DEBUG Write  0x001d: 12 80
2017-07-31 12:14:18,913 DEBUG Notify 0x001f: 12 80
2017-07-31 12:14:18,918 DEBUG Write  0x001d: 13 80
2017-07-31 12:14:19,115 DEBUG Notify 0x001f: 13 80 00 01
2017-07-31 12:14:19,120 DEBUG Write  0x001d: 14 80
2017-07-31 12:14:19,318 DEBUG Notify 0x001f: 14 80
2017-07-31 12:14:19,323 DEBUG Write  0x001d: 15 80
2017-07-31 12:14:19,520 DEBUG Notify 0x001f: 15 80
2017-07-31 12:14:19,525 DEBUG Write  0x001d: 16 80
2017-07-31 12:14:19,723 DEBUG Notify 0x001f: 16 80
2017-07-31 12:14:19,728 DEBUG Write  0x001d: 17 80
2017-07-31 12:14:19,925 DEBUG Notify 0x001f: 17 80
2017-07-31 12:14:19,930 DEBUG Write  0x001d: 18 80
2017-07-31 12:14:20,128 DEBUG Notify 0x001f: 18 80
2017-07-31 12:14:20,133 DEBUG Write  0x001d: 19 80
2017-07-31 12:14:20,331 DEBUG Notify 0x001f: 19 80
2017-07-31 12:14:20,336 DEBUG Write  0x001d: fe 80
2017-07-31 12:14:20,533 DEBUG Notify 0x001f: fe 80 00 01
2017-07-31 12:14:20,538 DEBUG Write  0x001d: 0b 84
2017-07-31 12:14:20,736 DEBUG Notify 0x001f: 0b 84 9d ba 2b 26 05
2017-07-31 12:14:20,741 DEBUG PyGattBackend, FC:3A:98:16:8A:0F initialized with status 0.
New client created: MetaWearClient, PyGattBackend, FC:3A:98:16:8A:0F, Model: 1, Firmware: 1.2.5
Get possible accelerometer settings...
{'data_rate in Hz': [0.78125, 1.5625, 3.125, 6.25, 12.5, 25.0, 50.0, 100.0, 200.0, 400.0, 800.0, 1600.0], 'data_range in Gs': ['16', '2', '4', '8']}
Write accelerometer settings...
2017-07-31 12:14:22,277 DEBUG Setting Accelerometer ODR to 3.125
2017-07-31 12:14:22,281 DEBUG Setting Accelerometer FSR to 4.0
2017-07-31 12:14:22,286 DEBUG Write  0x001d: 03 03 83 05
Check accelerometer settings...
data_rate in Hz: 3.125 data_range in Gs: 4.0
Subscribing to accelerometer signal notifications...
2017-07-31 12:14:23,358 DEBUG Subscribing to Accelerometer changes. (Sig#: 3047174144)
2017-07-31 12:14:23,363 DEBUG Write  0x001d: 03 04 01
2017-07-31 12:14:23,432 DEBUG Write  0x001d: 03 02 01 00
2017-07-31 12:14:23,495 DEBUG Write  0x001d: 03 01 01
Unsubscribe to notification...
2017-07-31 12:14:33,583 DEBUG Write  0x001d: 03 01 00
2017-07-31 12:14:33,654 DEBUG Write  0x001d: 03 02 00 01
2017-07-31 12:14:33,719 DEBUG Unsubscribing to Accelerometer changes. (Sig#: 3047174144)
2017-07-31 12:14:33,731 DEBUG Write  0x001d: 03 04 00
2017-07-31 12:14:38,806 DEBUG Write  0x001d: 09 08
2017-07-31 12:14:38,881 DEBUG Write  0x001d: 0a 05
2017-07-31 12:14:38,942 DEBUG Write  0x001d: 0b 0a

scaryghost, 2017-07-31:
Try restarting the Bluetooth device with the cleanup script and switch to the pybluez backend.

somacoder, 2017-07-31:
With pybluez, I cannot get a response from the device, even though I can with pygatt:

2017-07-31 21:35:03,611 DEBUG Added a <stdout> logging handler to logger: pymetawear
2017-07-31 21:35:03,614 INFO Creating MetaWearClient for FE:74:AE:2D:0B:E2...
2017-07-31 21:35:03,618 INFO Backend starter with pybluez for device address FE:74:AE:2D:0B:E2 with timeout None...
2017-07-31 21:35:03,620 INFO PyBluezBackend: Creating new GATTRequester...
2017-07-31 21:35:03,623 INFO PyBluezBackend: Connecting GATTRequester...
Traceback (most recent call last):
  File "acc.py", line 21, in <module>
    c = MetaWearClient(str(address), 'pybluez', debug=True)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.8.0-py2.7.egg/pymetawear/client.py", line 97, in __init__
    self.connect()
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.8.0-py2.7.egg/pymetawear/client.py", line 130, in connect
    self.backend.connect(clean_connect=clean_connect)
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.8.0-py2.7.egg/pymetawear/backends/pybluez.py", line 119, in connect
    super(PyBluezBackend, self).connect()
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.8.0-py2.7.egg/pymetawear/backends/__init__.py", line 92, in connect
    self._build_handle_dict()
  File "/usr/local/lib/python2.7/dist-packages/pymetawear-0.8.0-py2.7.egg/pymetawear/backends/pybluez.py", line 69, in _build_handle_dict
    for x in self.requester.discover_primary()}
RuntimeError: Device is not responding!

somacoder, 2017-07-31:
Fixed it. I went in through the iOS app and selected 'Update Firmware', even though it said the firmware was up to date. I then did a factory reset and now it works.

Sorry for the trouble!!

closed 2017-07-31 by somacoder