hbldh/pymetawear #42: libmetawear.so is not being built because of -Werror

Joshuaalbert, 2017-12-09:
Greetings,

I am attempting to install this package and upon attempting to import any of the code I get an error,
` /home/josh/git/pymetawear/pymetawear/libmetawear.so: cannot open shared object file: No such file or directory`
I installed by `python setup.py build`.
Actually, I notice there are some warnings that appear and you have the compiler flat `-Werror` set which treats warnings as an error.
After removing `-Werror` from `pymetawear/MetaWear-SDK-Cpp/config.mk` everything compiles.

If you would like a list of the warning I can send these.

## System
python 3.6 from conda, Fedora 25

hbldh, 2017-12-10:
Did this occur when building the `master` or the `develop` branch?

I ran into this myself, but I cannot replicate it now, even though I haven't made any changes regarding the issue... I believe it relates to this issue: https://github.com/mbientlab/MetaWear-SDK-Cpp/issues/14
Please add the warnings here as well so that we can verify that.

hbldh, 2018-04-25:
Since version 0.9.0, libmetawear is being built in the mbientlab/MetaWear-SDK-Python package. This is no longer an issue relating to PyMetWear directly.

closed 2018-04-25 by hbldh