hbldh/pymetawear #44: Ignore already subscribed sensorfusion notification

arturoribes, 2017-12-29:
Right now the sensorfusion module raises an exception if you try to subscribe to an already subscribed data signal. There is a use case (my use case) where this is not desirable:

Consider that I want to get the quaternion, but sometimes I also want acc/gyro data. I use a function like this:
```
    def enable_accgyro(self, enabled=True):
        if enabled:
            print("Subscribing to Sensor Fusion Quat/Acc/Gyro signal notifications...")
            self.c.sensorfusion.notifications(quaternion_callback=self.handle_quat,
                                              corrected_acc_callback=self.handle_acc,
                                              corrected_gyro_callback=self.handle_gyro)
        else:
            print("Subscribing to Sensor Fusion Quaternion signal notifications...")
            self.c.sensorfusion.notifications(quaternion_callback=self.handle_quat)
```

As you can see, if I have ```sensor.enable_accgyro(False)``` (which I do on startup), then, when I call it again like ```sensor.enable_accgyro(True)```, I get the exception at ```check_and_change_callback(self, data_signal, callback)``` from ```sensorfusion.py```. 

I think you should compare ```self._callbacks[data_signal][0]``` (which you don't use anywhere in your code) to the callback that gets passed to the function. However I have not been sucessful at that, so I solved it by commenting the condition that raises the exception.

hbldh, 2018-01-31:
I am sorry for not getting back to you regarding this issue earlier; I was in the process of rewriting the pymetawear quite a bit (can be seen partially in the develop branch), but I got preoccupied with other things while waiting for a PR to go through.

I did not write the sensor fusion module myself, so I am not quite certain why it is the way it is now. Is the MetaWear board actually capable of providing several sensor fusion signals, and even if it can, is the data synced and trustworthy? Have you verified this on the Mbientlab community board or with anyone working there? I have not seen this capability in other drivers, but to be honest I haven't looked that closely either...

I will get back to PyMetaWear during February I think, and the `sensorfusion` module is one of the things I will look at now that I actually have a MetaWear-board that can give me such data. I will take your request with me.

hbldh, 2018-02-04:
@arturoribes Will be releasing a new version including this fix shortly. Need to test it a bit more first.

hbldh, 2018-04-25:
This was released in Release v0.9.1, commit 4d681a72a.

closed 2018-04-25 by hbldh