hbldh/pymetawear #50: Sensor fusion callback only working intermittent?

how-chen, 2018-06-15:
I am using one of the metawear R sensors with an Ubuntu laptop (16.04) with pymetawear. The accelerometer and gyroscope examples are working fine at the moment, but the sensor fusion example is only working on an intermittent basis (more often than not, it is problematic). I can run the code a few times and everything is working then the code will run, but not streaming any data (I can connect to the sensor but the callback does not seem to be returning anything). I am out of ideas. Was wondering if anyone has run into the same problem.

hbldh, 2018-06-15:
The board you have, it is a MetaMotion R board? You are sure that it supports sensor fusion data?
I have not experienced this particular problem, but I know that repeatedly connecting to the board might cause lockups if the connections are not terminated properly.

MetaWear has recently released a Python BLE package of its own, and I plan to upgrade pymetawear to use it. Maybe that will alleviate the problem.

hbldh, 2018-06-18:
@how-chen I will make some investigations into the matter during this week. Will get back to you on this.

hbldh, 2018-08-10:
@how-chen If you are still interested in trying to resolve this issue, I have just released a new version (v0.11.0) of metawear which uses a different BLE library for communication. This might resolve your issue.

bgromov, 2018-08-14:
@how-chen I have seen these issues with my MetaWear R+ sensors on Linux with a SpeedLink dongle, but never had any problems on macOS on a macbook. When there are two sensors sometimes it hicks up also on macOS, but with a single sensor everything works perfectly.

Also, have a look at this [thread](https://mbientlab.com/community/discussion/1850/sensor-stops-sending-data).

As a wild guess: it seems that the throughput of a particular BLE chip may affect this behaviour, i.e. if the BLE unable to deliver data within certain time the sensor gives up. So I would recommend to try to play with the parameters for sensor fusion sampling rates AND connection parameters.

I have the following setup and everything works fine on macOS with a single sensor:

``` python
client.settings.set_connection_parameters(min_conn_interval=7.5, max_conn_interval=10, latency=0, timeout=3000)

client.sensorfusion.set_mode(SensorFusionMode.NDOF)
client.sensorfusion.set_acc_range(SensorFusionAccRange._16G)
client.sensorfusion.set_gyro_range(SensorFusionGyroRange._2000DPS)

client.sensorfusion.set_sample_delay(SensorFusionData.QUATERNION, 10)
client.sensorfusion.set_sample_delay(SensorFusionData.CORRECTED_ACC, 10)
client.sensorfusion.set_sample_delay(SensorFusionData.CORRECTED_GYRO, 10)
```

That's all with the latest `pymetawear` version (v0.11.0)

how-chen, 2018-08-15:
@hbldh Thanks for the follow up. I should get around to it this weekend. @bgromov Glad I'm not the only one with this problem. I'm an a native ubuntu machine (Carbon X1 gen 2) using its internal BLE connection.