hbldh/pymetawear #51: Support metawear v0.4.0

qutang, 2018-06-19:
This version adds support for windows 10. Could be very useful for many developers using windows. 

https://github.com/mbientlab/MetaWear-SDK-Python/commit/1a7e92ca0bb46c6f814e387c96cd4aa8d6bfcd53

hbldh, 2018-06-20:
Definitely! I have seen the possibility but I have not had time to deal with it. It requires a fair bit of changes in pymetawear and even more tests to make me able to release a version using 0.4.0, but it will happen.
If you feel up to the challenge, feel free to attempt the upgrade!

hbldh, 2018-08-08:
@qutang Released `pymetawear` version 0.11.0 which uses `metawear` version 0.5.0. `pymetawear` can therefore now be used from Windows platforms.

closed 2018-08-08 by hbldh