hbldh/pymetawear #56: Error at "building wheel" when installing

thoraxmax, 2018-09-13:
* pymetawear version: most recent
* Python version: 3.7
* Operating System: Windows 10

### Description

I tried to install the build with pip in cmd run as admin, I got a build error when "building wheel". Something with the MetaWear.Win32.dll file. 

### What I Did
I have installed VS2017 community, the win to sdk version, warble and pywarble without errors. Now suddenly can't get any further. than the errors from the command in cmd. 

Help is very much appreciated, I am new to programming but I am studying it a lot right now. I am very exited to use python with my metamotion! 
```
C:\WINDOWS\system32>py -m pip install pymetawear
Collecting pymetawear
  Using cached https://files.pythonhosted.org/packages/c9/cc/dbc0eede0db795848ffd3b51a170a818df40420466b256361a66b9281e1c/pymetawear-0.11.0-py2.py3-none-any.whl
Requirement already satisfied: tqdm in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from pymetawear) (4.25.0)
Collecting metawear==0.5.0 (from pymetawear)
  Using cached https://files.pythonhosted.org/packages/cc/47/ace0feea8cb880b4dfe4b7a2099a0b89cc9a76acdaa23ea87d25bbcdc2ff/metawear-0.5.0.tar.gz
Requirement already satisfied: warble<2.0,>=1.1 in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from metawear==0.5.0->pymetawear) (1.1.0)
Requirement already satisfied: requests in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from metawear==0.5.0->pymetawear) (2.19.1)
Requirement already satisfied: certifi>=2017.4.17 in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from requests->metawear==0.5.0->pymetawear) (2018.8.24)
Requirement already satisfied: urllib3<1.24,>=1.21.1 in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from requests->metawear==0.5.0->pymetawear) (1.23)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from requests->metawear==0.5.0->pymetawear) (3.0.4)
Requirement already satisfied: idna<2.8,>=2.5 in c:\users\thors\appdata\local\programs\python\python37-32\lib\site-packages (from requests->metawear==0.5.0->pymetawear) (2.7)
Building wheels for collected packages: metawear
  Running setup.py bdist_wheel for metawear ... error
  Complete output from command C:\Users\thors\AppData\Local\Programs\Python\Python37-32\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\thors\\AppData\\Local\\Temp\\pip-install-th0eyhsa\\metawear\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d C:\Users\thors\AppData\Local\Temp\pip-wheel-ug99r3st --python-tag cp37:
  running bdist_wheel
  running build
  running build_py
  Microsoft (R) Build Engine version 15.8.168+ga8fba1ebd7 for .NET Framework
  Copyright (C) Microsoft Corporation. All rights reserved.

  Build started 2018-09-13 18:29:08.
  Project "C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj" on node 1 (default targets).
  PrepareForBuild:
    Creating directory "build\Win32\Release\MetaWear.Win32\".
  C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(399,5): warning MSB8029: The Intermediate directory or Output directory cannot reside under the Temporary directory as it could lead to issues with incremental build. [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
    Creating directory "dist\Release\lib\Win32\".
    Creating directory "build\Win32\Release\MetaWear.Win32\MetaWear.Win32.tlog\".
  InitializeBuildStatus:
    Creating "build\Win32\Release\MetaWear.Win32\MetaWear.Win32.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
  ClCompile:
    C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\HostX86\x86\CL.exe /c /Zi /nologo /W3 /WX- /diagnostics:classic /O2 /Oi /Oy- /GL /D METAWEAR_DLL_EXPORTS /D _WINDLL /D _UNICODE /D UNICODE /Gm- /EHsc /MD /GS /Gy /fp:precise /Zc:wchar_t /Zc:forScope /Zc:inline /Fo"build\Win32\Release\MetaWear.Win32\\" /Fd"build\Win32\Release\MetaWear.Win32\vc141.pdb" /Gd /TP /analyze- /FC /errorReport:queue src\metawear\core\cpp\anonymous_datasignal.cpp src\metawear\core\cpp\datasignal.cpp src\metawear\core\cpp\debug.cpp src\metawear\core\cpp\event.cpp src\metawear\core\cpp\logging.cpp src\metawear\core\cpp\macro.cpp src\metawear\core\cpp\moduleinfo.cpp src\metawear\core\cpp\responseheader.cpp src\metawear\core\cpp\settings.cpp src\metawear\core\cpp\timer.cpp src\metawear\core\cpp\version.cpp src\metawear\dfu\cpp\dfu_operations.cpp src\metawear\dfu\cpp\dfu_operations_details.cpp src\metawear\dfu\cpp\dfu_utility.cpp src\metawear\dfu\cpp\file_operations.cpp src\metawear\dfu\cpp\miniz.cpp src\metawear\impl\cpp\datainterpreter.cpp src\metawear\impl\cpp\metawearboard.cpp src\metawear\peripheral\cpp\haptic.cpp src\metawear\peripheral\cpp\ibeacon.cpp src\metawear\peripheral\cpp\led.cpp src\metawear\peripheral\cpp\neopixel.cpp src\metawear\platform\cpp\async_creator.cpp src\metawear\platform\cpp\memory.cpp src\metawear\platform\cpp\task.cpp src\metawear\platform\cpp\threadpool.cpp src\metawear\processor\cpp\dataprocessor.cpp src\metawear\processor\cpp\dataprocessor_config.cpp src\metawear\sensor\cpp\accelerometer.cpp src\metawear\sensor\cpp\accelerometer_bosch.cpp src\metawear\sensor\cpp\accelerometer_mma8452q.cpp src\metawear\sensor\cpp\ambientlight_ltr329.cpp src\metawear\sensor\cpp\barometer_bosch.cpp src\metawear\sensor\cpp\colordetector_tcs34725.cpp src\metawear\sensor\cpp\conductance.cpp src\metawear\sensor\cpp\gpio.cpp src\metawear\sensor\cpp\gyro_bmi160.cpp src\metawear\sensor\cpp\humidity_bme280.cpp src\metawear\sensor\cpp\magnetometer_bmm150.cpp src\metawear\sensor\cpp\multichanneltemperature.cpp src\metawear\sensor\cpp\proximity_tsl2671.cpp src\metawear\sensor\cpp\sensor_fusion.cpp src\metawear\sensor\cpp\serialpassthrough.cpp src\metawear\sensor\cpp\switch.cpp src\metawear\sensor\cpp\utils.cpp
    anonymous_datasignal.cpp
    datasignal.cpp
    debug.cpp
    event.cpp
    logging.cpp
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\core\cpp\logging.cpp(318): warning C4244: 'initializing': conversion from 'float' to 'uint32_t', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\core\cpp\logging.cpp(674): warning C4244: '=': conversion from 'double' to 'float', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
    macro.cpp
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\core\cpp\macro.cpp(41): warning C4267: 'argument': conversion from 'size_t' to 'uint8_t', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
    moduleinfo.cpp
    responseheader.cpp
    settings.cpp
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\core\cpp\settings.cpp(89): warning C4244: '/=': conversion from 'float' to 'uint16_t', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
    timer.cpp
    version.cpp
    dfu_operations.cpp
    dfu_operations_details.cpp
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\dfu\cpp\dfu_operations_details.cpp(76): warning C4244: 'initializing': conversion from 'double' to 'int', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
    dfu_utility.cpp
    file_operations.cpp
  c:\users\thors\appdata\local\temp\pip-install-th0eyhsa\metawear\metawear-sdk-cpp\src\metawear\dfu\cpp\file_operations.cpp(43): warning C4244: '=': conversion from 'double' to 'int', possible loss of data [C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\MetaWear-SDK-Cpp\MetaWear.Win32.vcxproj]
[... cut ...]
Command "C:\Users\thors\AppData\Local\Programs\Python\Python37-32\python.exe -u -c "import setuptools, tokenize;__file__='C:\\Users\\thors\\AppData\\Local\\Temp\\pip-install-th0eyhsa\\metawear\\setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record C:\Users\thors\AppData\Local\Temp\pip-record-cilr31n9\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in C:\Users\thors\AppData\Local\Temp\pip-install-th0eyhsa\metawear\

C:\WINDOWS\system32>
```

qutang, 2019-09-24:
the metawear package version should be bumped to 0.7.0 to solve this problem. Unfortunately the author seems not maintaining the repository any more.

hbldh, 2020-03-10:
This is solved in the v0.12.0 release.

closed 2020-03-10 by hbldh