hbldh/pymetawear #66: Segmentation fault when connecting to an already connected device

ironss, 2020-10-22:
If you call connect() for a MetaWearClient when it is already connected, Python dumps core.

I would expect some sort of clean failure, presumably an exception.

The underlying C++ library seems to throw an exception 'BLEPP::SocketConnectFailed'; Python does not handle it correctly.

If the existing MetaWearClient is disconnected, then the new MetaWearClient works correctly

```
$ .venv/bin/python3
>>> import pymetawear.client
>>> addr = 'EB:02:78:03:58:9D'
>>> dev = pymetawear.client.MetaWearClient(addr)
error 1603337254.958039: Error on line: 296 (src/blestatemachine.cc): Operation now in progress
Connected
Services disconvered
Characteristics discovered
Descriptors found
>>> dev.connect()
error 1603338727.078311: Error on line: 296 (src/blestatemachine.cc): Device or resource busy
terminate called after throwing an instance of 'BLEPP::SocketConnectFailed'
  what():  Device or resource busy
Aborted (core dumped)
```

Versions
pymetawear: 0.12.0
Python: python3
OS: Ubuntu 20.04

$ .venv/bin/pip3 freeze
certifi==2020.6.20
chardet==3.0.4
idna==2.10
metawear==0.7.0
pymetawear==0.12.0
requests==2.24.0
tqdm==4.50.2
urllib3==1.25.11
warble==1.2.0

ironss, 2020-10-22:
By juggling an ```_is_connected flag```, you could work around this issue. However, the problem is in the underlying library, and there is a situation where the ```_is_connected``` flag will not work: if you create a _new_ MetaWear client pointing at the same underlying device:

```
>>> dev1 = pymetawear.client.MetaWearClient(addr)
error 1603337254.958039: Error on line: 296 (src/blestatemachine.cc): Operation now in progress
Connected
Services disconvered
Characteristics discovered
Descriptors found
>>> dev2 = pymetawear.client.MetaWearClient(addr)
error 1603338727.078311: Error on line: 296 (src/blestatemachine.cc): Device or resource busy
terminate called after throwing an instance of 'BLEPP::SocketConnectFailed'
  what():  Device or resource busy
Aborted (core dumped)
```
This might happen if your program presents a list of devices, and the user selects one that is already connected.

hbldh, 2020-10-22:
I think that is better fixed in the other libraries. Adding it here makes less sense, imo. Rewrite a minimal example using the `metawear` package (https://github.com/mbientlab/MetaWear-SDK-Python) and ask them to integrate it there.

closed 2023-06-05 by hbldh