hbldh/pymetawear #74: Can't import 'FnVoid_DataP' from 'mbientlab.metawear.cbindings'

ZiruiHong, 2023-05-30:
* pymetawear version: 0.12.0
* Python version: 3.7
* Operating System: ubuntu 20.04

### Description

I'm trying to run metawear_node.py in ROS (noetic version) hosted by a Raspberry Pi 4. An import error came up saying "cannot import name 'FnVoid_DataP' from 'mbientlab.metawear.cbindings'". I paste the error message as below.

```
ubuntu@ubuntu:~/lciii_ws$ rosrun metawear_ros metawear_node.py 
Traceback (most recent call last):
  File "/home/ubuntu/lciii_ws/src/metawear_ros-master/scripts/metawear_node.py", line 9, in <module>
    from pymetawear.client import MetaWearClient
  File "/usr/local/lib/python3.7/dist-packages/pymetawear/client.py", line 22, in <module>
    from pymetawear import add_stream_logger, modules
  File "/usr/local/lib/python3.7/dist-packages/pymetawear/modules/__init__.py", line 16, in <module>
    from .base import PyMetaWearModule, Modules
  File "/usr/local/lib/python3.7/dist-packages/pymetawear/modules/base.py", line 22, in <module>
    from mbientlab.metawear.cbindings import FnVoid_DataP, DataTypeId, \
ImportError: cannot import name 'FnVoid_DataP' from 'mbientlab.metawear.cbindings' (/usr/local/lib/python3.7/dist-packages/mbientlab/metawear/cbindings.py)
```

### What I Did
I checked the metawear cbindings.py file and found there was no class or variable called FnVoid_DataP (see the screenshot below). I noticed that this file was generated by MetaWear C++ API. So I tried to re-install the Metawear python package but it still didn't work. 

[image]

Hope you can share some clues to help me address the issue.

hbldh, 2023-05-30:
This library is no longer up to date with Metawear's releases. The latest release is tailored for metawear==0.7.0, everything else will probably not work. I can not provide any meaningful 

I kind of lost interest in this library when they dropped Windows support. I started to integrate it with my other BLE library [hbldh/bleak](https://github.com/hbldh/bleak), but when the actual Metawear library became so hard to build on Windows it was no longer reasonable to keep working on this.

The `FnVoid`... must have changed. It seems version 0.12.1a1 that is present in the master branch does not have that specific import. Try installing
```
pip install git+https://github.com/hbldh/pymetawear.git#egg=pymetawear
```
and see if that works better. That is all I have to offer right now.

ZiruiHong, 2023-05-30:
I see. Thanks for your quick response. 

Best wishes