hbldh/pymetawear #75: Fails to build under Debian 12

kalsan, 2024-07-08:
* pymetawear version: 0.12.0
* Python version: 3.12.3 (miniconda)
* Operating System: Debian 12

### Description

After installing all listed dependencies via apt, when running `pip install pymetawear`, the build process crashes as follows:

```
g++ -MMD -MP -MF "build/x64/release/src/metawear/sensor/cpp/magnetometer_bmm150.d" -c -o build/x64/release/src/metawear/sensor/cpp/magnetometer_bmm150.o -std=c++11 -fPIC -fvisibility=hidden -fvisibility-inlines-hidden -Wall -Werror -Isrc -DMETAWEAR_DLL -DMETAWEAR_DLL_EXPORTS -Wno-strict-aliasing -O3 -m64 src/metawear/sensor/cpp/magnetometer_bmm150.cpp
      src/metawear/sensor/cpp/utils.cpp: In function ‘char* copy_string(const char*)’:
      src/metawear/sensor/cpp/utils.cpp:26:11: error: ‘char* strcpy(char*, const char*)’ writing one too many bytes into a region of a size that depends on ‘strlen’ [-Werror=stringop-overflow=]
         26 |     strcpy(buffer, src);
            |     ~~~~~~^~~~~~~~~~~~~
      src/metawear/sensor/cpp/utils.cpp:25:38: note: destination object of size [0, 9223372036854775805] allocated by ‘malloc’
         25 |     char* buffer = (char*)std::malloc(strlen(src));
            |                           ~~~~~~~~~~~^~~~~~~~~~~~~
      cc1plus: all warnings being treated as errors
      make: *** [Makefile:69: build/x64/release/src/metawear/sensor/cpp/utils.o] Fehler 1
      make: *** Es wird auf noch nicht beendete Prozesse gewartet....
      make: Verzeichnis „/tmp/pip-install-15wo1e71/metawear_a1ec0788994347e3bf9320e005dd15d5/MetaWear-SDK-Cpp“ wird verlassen
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-install-15wo1e71/metawear_a1ec0788994347e3bf9320e005dd15d5/setup.py", line 66, in <module>
          setup(
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/__init__.py", line 104, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 184, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 200, in run_commands
          dist.run_commands()
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
          super().run_command(command)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/me/miniconda3/lib/python3.12/site-packages/wheel/bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
          super().run_command(command)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/command/build.py", line 132, in run
          self.run_command(cmd_name)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/dist.py", line 967, in run_command
          super().run_command(command)
        File "/home/me/miniconda3/lib/python3.12/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-install-15wo1e71/metawear_a1ec0788994347e3bf9320e005dd15d5/setup.py", line 56, in run
          raise RuntimeError("Failed to compile C++ SDK")
      RuntimeError: Failed to compile C++ SDK
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for metawear
  Running setup.py clean for metawear
Failed to build metawear
ERROR: Could not build wheels for metawear, which is required to install pyproject.toml-based projects
```

hbldh, 2024-07-08:
I have not maintained this library at all for a long time, and I doubt it works with any recent metawear boards. Use their official Python sdk instead of this.

kalsan, 2024-07-09:
Thanks for your quick answer. They have stopped development and support as well and are going mobile-only. I suppose this is the end of Metawear on Desktop.